Jean-MarcHarvengt/VGA_t4 #1: Teensy 4.0 Support?

YorkFieldX, 2020-10-27:
Hi,

I'm experimenting around with VGA on Teensy 4.0, and came across your project. Great work!
However, I've noticed you have pin 33 used, which is obviously missing on Teensy 4.0. I'm really new to this, and I didn't find any examples on how to change pins in code. Could you please advice how to get this running on Teensy 4.0?

Jean-MarcHarvengt, 2020-10-27:
Hi,

Pin 33 is there on Teensy4.0. It is underneath, one of the 10 tracks, the row at the level of VBAT.
In fact I initially developed the VGA driver for the Teensy4.0.
If you have a Teensy 4.0 without pins yet,  you can use longer header pins (one pin more on each side) and wire the pin 33 underneath to one of the 2 extra pins.

I did not try the audio driver but should also work.
T4.0 is only tricky for SD card and you won't be able to wire PSRAM and SDcard. You will need pins 34 to 39 for both so you can only connect SDcard (also smaller tracks underneath)

Good luck!

YorkFieldX, 2020-10-28:
Hi, 
Thank you for fast response! I'll try that tonight. In my use case there's no need for an SD card or audio since image is generated locally on Teensy (image data received from another device over 8bit parallel port).
Also, could you please advice if it's possible to simplify the circuit if only output colours used are black/white?
I see how 2 of the VGA Colour Pins require 3 resistors, but the other one requires only 2.
Thank you once again for your help!

Jean-MarcHarvengt, 2020-10-28:
I use 8 bits to store RGB data, 3bits for R, 3bits for G and 2 for B. You create 3 simple DACs on 3, 3 and 2 bits.
If you only want B/W you can just use 3 pins.  2 pins for H and V sync, 1 for B/W (0/1). You have to play with the shift register to output only 1 bit instead of 4  (2 shift registers => 2x4). You can use a single DMA and a single shift register for that.

YorkFieldX, 2020-11-16:
Hi!
Could you by any chance share the photos of the wiring please? I've had no luck getting it to work on T4.0, so I got a T4.1, yet running into some issues (tv shows no signal). I believe I followed the instructions in 'wiring' section of readme correctly, however I'm really new to all of this so I possibly made a mistake somewhere.

I did a breakout of VGA connector (split cable in half) onto a breadboard (pins 1-15).
For connections with resistors on them, I took some jumper wires, cut them in half, and soldered appropriate resistors in. What really confuses me is that in 'schematics' files I could see pins corresponding to colours (VGA 1, 2 & 3) shorted to ground via resistors, but in 'wiring' section it's more like
Teensy Pin A --> Resistor A --> VGA Pin 1
Teensy Pin B --> Resistor B --> VGA Pin 1
Teensy Pin C --> Resistor C --> VGA Pin 1

Am I missing something?

Right now the issue is TV is showing "No Signal". Interestingly, if VGA Pin 15 is not shorted to ground it won't even recognise there's a VGA device connected.

I've double checked the breakout of the VGA plug and breadboard wiring, everything seems fine according to my understanding of the 'wiring' section.
Sorry to bother you and really appreciate your help!

Jean-MarcHarvengt, 2020-11-16:
if you look in the GitHub sources of the project, there is in schematics dir  the file schematic_T4VGA.png.

That should be readable.
There is no shorted to ground be careful  !!!

5,6 and 25 all go to RED via resistors.
4,13 and 35 all go to GREEN via resistors.
12 and 14 all go to BLUE via resistors

10 goes to V-SYNC via resistor
37 goes to H-SYNC via resistor

Don't forget to ground the pins 5,10,5,7,8 at the VGA connector.

All the rest you don't need to test VGA....

YorkFieldX, 2020-11-23:
Thank you for your help! I finally managed to get it to work :)

closed 2020-11-23 by YorkFieldX