Jean-MarcHarvengt/VGA_t4 #3: question about the connection

DaanWesterhof, 2021-03-29:
Hello,

I'm realy exited to get vga working with my teensy4.1 but i am a bit confused regarding the resistors and connections. The resistor values specified in the readme are difrent from the values in the schematic. Wich should i follow. Im pretty sure i have to connect all 3 red 3 green and 2 blue pins like in the schematic but the readme makes it seem like i could also connect only 1 per colour. 

So should i follow the schematic, or follow the readme? And if i have to follow the readme do i need to connect all 3/2 pins per colour or only 1?

kind regards,

Daan

Jean-MarcHarvengt, 2021-03-30:
Hi,

3 pins are used for red, 3 for green and 2 for blue.
For red, the 3 pins are connected to the red of the VGA via a resistor, one per pin. Same principle for green and blue. Next to that the hsync and vsync are connected to the hsync and vsync of the VGA via a single resistor each. The principle is similar to the great uVGA lib for the teensy 3.6. you can Google for it, only the pins are different!   so readme is probably easier to understand, sorry. J-M this is the simplest way. You can also use full resistor ladder but it is not really needed.

DaanWesterhof, 2021-03-30:
Hey,

Thanks for the reply, its all clear now. Im going to try and make a pcb breakout for the teensy so i can solder the vga connector on there, so thank you very much for your reply, itl help out a lot!

closed 2021-03-30 by DaanWesterhof