Jean-MarcHarvengt/VGA_t4 #4: What if i pull out the vga cable?

DaanWesterhof, 2021-03-31:
Hey
While designing my pcb i tought of something. If i have the teensy turned on but pull out the vga cable while its running, wont that be bad for the output ports? As they might damage themselfs if one or two are on high output and one or two on low. 

I was thinking about using pull down resistors to protec the system when this happens. As i am not sure if the internal pull down resistors are already activated. But i think that that also might have impact on the color values. 

So i was wondering if there was already protection in place for this problem or if using the schematic from darthvader would add this protection.

I hope you can help me,

Kind Regards,

Daan

Jean-MarcHarvengt, 2021-04-03:
sorry for the late reply...
I am not an expert but that should not be an issue for output pins. 
VGA signal just goes out.
A guy created an EGA to VGA adapter based on the Teensy, there he had to buffer the inputs and be careful with ESD but not for outputs.

Regards,

J-M

DaanWesterhof, 2021-04-07:
Alright, thanks a lot!

I managed to get it working with a breadboard for now and am pretty happy with the results so far. Now I can finish up my pcb and then I can finally work without a mess of jumper wires and resistors. Thanks a lot for providing this library!

Kind regards,

Daan

closed 2021-04-07 by DaanWesterhof