Jean-MarcHarvengt/VGA_t4 #5: SPI2 SD card interfering wiht VGA sync

gertk64, 2021-04-16:
Hi, 
First I must say, nice job in getting a stable VGA output from the Teensy (I am using a 4.0)

But I managed to get myself into problems by using the SD card SPI2 (the pads at the underside since the pins for SPI1 are unavailable)
 
Every time I access the SD card the VGA output goes out of sync, luckily it restores itself after the access is done.
I skimmed through the source of the SPI library but it looks like it either:

disables interrupts
or
interferes with the IO pins

I also have an USB keyboard attached to the host port pins and this works fine.

Any ideas how to solve this ? 

Thanks in advance,
Gert

Jean-MarcHarvengt, 2021-04-21:
hi Gert,

Any extra DMA use will interfere with the VGA rendering.
This is because the driver is already using 2 DMAs to render the pixels of each scanline.

I feel sorry about that...
I had same issue for audio so I used interrupt based I2S transfer.

Normally you can also use SPI in non DMA mode.
That should be better however you might lower the speed.
Have you tried that?
I have used that for controlling LCD but not with SD card.
you could try controlling first an ILI display with SPI in SW mode to see if any interference is visible?

Regards,

J-M

gertk64, 2021-04-22:
Hi J-M,
Thanks for the answer, yes I tried to modify the settings for the SD card interface for software SPI but keep getting the message that it misses kinetis.h or such. Anyway that is not your problem :-) Will try to dig further.

BTW: the synchronization of those 2 DMA channels is the pixel offset defined by the hardware? I found in the enable DMA bit selection that you can start both DMA channels at the same time but it makes no difference in the output of the pixel streams so I guess the difference is in the access time internally since both DMA channels need to read from memory. I have not much experience that deep into this ARM CPU yet. Are there separate RAM banks in which DMA access can be concurrently ?

This seems to work but no difference in pixel shift:

`// Enable DMAs
    // DMA_SERQ = flexio2DMA.channel;
    // DMA_SERQ = flexio1DMA.channel;
    DMA_SERQ = ( flexio2DMA.channel | flexio1DMA.channel | 0x40 );  // in one go ?
`

Regards,
Gert

closed 2021-04-22 by gertk64