Jean-MarcHarvengt/VGA_t4 #7: Trying to use VGA_t4 and 3rd I2S port with Teensy 4.1

lerabot, 2022-01-31:
First of all, thanks for you library, it's quite nice :)
I'm currently making a wierd arcade using your library but I have trouble adding a I2S device to my project.

I know that I can't use pin 16->19 because of flexIO, but I assumed I could use pin 24-25 (SCL2, SDA2).
I tested the device and I can get it to work, as soon as I turn on the vga.begin() I get erratic behavior.
The teensy will reset after 15-20 second and my I2S communication (confirmed to be on SCL2 & SDA2) is broken. 

Any clues?

Jean-MarcHarvengt, 2022-02-01:
hi, I2S is used for audio in the vga driver once you call begin_audio, I assume you meant I2C? There are few I2C port on the teensy. Have you tried others? It might be also that you need to write your own I2C driver, interrupt based to check what goes wrong. Are you using I2C slave or master? Again I assume you meant I2C, I2S is used in the library for audio and there it is used in the MCUME project in all emulator.

lerabot, 2022-02-02:
You are completely right! I meant I2C, my bad.
I'm trying to integrate these little things -> https://learn.adafruit.com/neokey-1x4-qt-i2c/arduino
I have a feeling they're slave devices.